ros2/rosidl_python #5: Memory leaks in rosidl python generated code

martins-mozeiko, 2018-06-29:
I have found that Python generated code for message type conversion leaks a lot of memory. This happens inside convert_from_py functions for messages that have non-primitive types as fields. convert_from_py function is called both for every message that is sent and received (rclpy_publish and rclpy_take functions from _rclpy.c file).

For example, if message X contains "std_msgs/Header header" as field, the following fragment is generated in its X_convert_from_py function:
```
  {  // header
    ...
    ... omitting irrelevant code to get convert_from_py function ...
    ...
    std_msgs__msg__Header * tmp = (std_msgs__msg__Header *) convert_from_py(field);
    if (!tmp) {
      return NULL;
    }
    ros_message->header = *tmp;
    Py_DECREF(field);
  }
```
Nobody frees tmp pointer which is malloc'ed when calling convert_from_py function pointer (for header type).

The question is here is how to free tmp pointer?

* it is wrong to  `destroy_from_py(tmp)` function before Py_DECREF, because caller of X_convert_from_py (like rclpy_publish from _rclpy.c) will call destroy_ros_message on object, and that will destroy ros_message->header members. Header contains String type which means string destroy function will be called twice and this will either crash program or corrupt memory.

* it is wrong to put just `free(tmp)` there, because it won't release the objects inside tmp.

One potential solution would be to introduce new convert_from_py function for each type. Something like convert_from_py_internal.

Its signature would be: `void convert_from_py_internal(PyObject* obj, void* msg);`
Its purpose would be not to allocate any memory, but only assign necessary fields to existing message structure.

This way code snippet above would change to this:
```
  {  // header
    ...
    ... omitting irrevant code to get convert_from_py_internal function ...
    ...
    convert_from_py_internal(field, &ros_message->header);
    Py_DECREF(field);
  }
```

The callers (like rclpy_publish function in _rclpy) would still could call convert_from_py to get newly allocated C message structure, but internal conversions would happen directly on fields of already allocated message structures - no extra allocations, and no memory leaks.

What do you think about this? I can create pull request that implements this.
Let me know if you need more information on this.

martins-mozeiko, 2018-06-30:
Here's the change I am proposing: https://github.com/martins-mozeiko/rosidl_python/commit/f4edf57ed8bfd9fb3a68c5edf9e5644bfa697c80
Currently I am still testing it, but it seems to work fine.

martins-mozeiko, 2018-07-06:
I implemented better solution - it splits memory allocation from conversion functionality.

This way only top level caller needs to allocate message when it wants to put data into message. Inner calls (for nested fields) simply call conversion function without extra memory allocation for nested fields (which leaked).

This also allowed to improve how rclpy_take and rclpy_take_request function did conversion.

Before they created empty Python message, then converted to C message, then called actual rcl_take or rcl_take_request that fills C message and then converted again to new Python message which is returned to caller.

Now they only allocate empty C message, calls rcl_take/rcl_take_request and then coverts to new Python message that is returned.

This is faster and uses less memory. I also fixed couple more potential leaks where reference count was not decreased or message not destroyed - especially in error cases.

These modifications are done in two repositories - rosidl_python and rclpy. They cannot be used independently, changes in them depend on other repo.

https://github.com/martins-mozeiko/rosidl_python/commit/3443d1cdd32b3d09deb4d2d08e18bbf63b449416
https://github.com/martins-mozeiko/rclpy/commit/303b4f25f68470f816eb0c3a9d3da7ea84c0b2dc

dirk-thomas, 2018-07-10:
Sorry, for the delay in feedback. We had to "recover" from the Bouncy release week :wink:

The overall patch looks good to me. I added a few minor comments on the [commit](https://github.com/martins-mozeiko/rosidl_python/commit/f4edf57ed8bfd9fb3a68c5edf9e5644bfa697c80). Please create a PR for it.

dirk-thomas, 2018-07-10:
The second approach is certainly even better. I added only minor comments there too. Please create a PR for the two commits of the second approach.

dirk-thomas, 2018-07-12:
I tried the referenced patches and they fix the memory leak for me :tada:

A set of CI builds to check for unexpected regressions:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

dirk-thomas, 2018-07-12:
The failing linter tests seem to be the ones addressed in ros2/rosidl_python#6. The forked repo with the PR isn't up-to-date. After merging the patches this shouldn't be a problem on master.

dirk-thomas, 2018-07-12:
Actually some linter warnings were unrelated to that. I addressed them in https://github.com/ros2/rclpy/pull/203/commits/e6275fa02244e9191d7590e1131c05c1bbea91ce: [image]

dirk-thomas, 2018-07-12:
@martins-mozeiko Thank you for working on this significant fix!

closed 2018-07-12 by dirk-thomas