ros2/rosidl_python #16: Add support to get message field types

fkromer, 2018-10-29:
In ROS1 it's possible to get the message fields (here: `'__slots__': ['x', 'y', 'z']`) and their corresponding types (here: `'_slot_types': ['float64', 'float64', 'float64']` via `_get_types()`) from messages (here: `geometry_msgs.msg.Point`).

```
$ ipython
Python 2.7.6 (default, Nov 23 2017, 15:49:48) 
Type "copyright", "credits" or "license" for more information.

IPython 5.5.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: from geometry_msgs.msg import Point

In [4]: Point.__dict__
Out[4]: 
dict_proxy({'__doc__': None,
            '__init__': <function geometry_msgs.msg._Point.__init__>,
            '__module__': 'geometry_msgs.msg._Point',
            '__slots__': ['x', 'y', 'z'],
            '_full_text': '# This contains the position of a point in free space\nfloat64 x\nfloat64 y\nfloat64 z\n',
            '_get_types': <function geometry_msgs.msg._Point._get_types>,
            '_has_header': False,
            '_md5sum': '4a842b65f413084dc2b10fb484ea7f17',
            '_slot_types': ['float64', 'float64', 'float64'],
            '_type': 'geometry_msgs/Point',
            'deserialize': <function geometry_msgs.msg._Point.deserialize>,
            'deserialize_numpy': <function geometry_msgs.msg._Point.deserialize_numpy>,
            'serialize': <function geometry_msgs.msg._Point.serialize>,
            'serialize_numpy': <function geometry_msgs.msg._Point.serialize_numpy>,
            'x': <member 'x' of 'Point' objects>,
            'y': <member 'y' of 'Point' objects>,
            'z': <member 'z' of 'Point' objects>})
$ python
>>> from geometry_msgs.msg import Point
>>> p = Point()
>>> p.__slots__
['x', 'y', 'z']
>>> p._get_types()
['float64', 'float64', 'float64']
```

ROS2 doesn't support this right now.

```
$ ipython3
Python 3.6.5 (default, Apr  1 2018, 05:46:30) 
Type "copyright", "credits" or "license" for more information.

IPython 5.5.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: from geometry_msgs.msg import Point

In [2]: Point.__dict__
Out[2]: 
mappingproxy({'__doc__': "Message class 'Point'.",
              '__eq__': <function geometry_msgs.msg._point.Point.__eq__>,
              '__hash__': None,
              '__init__': <function geometry_msgs.msg._point.Point.__init__>,
              '__module__': 'geometry_msgs.msg._point',
              '__repr__': <function geometry_msgs.msg._point.Point.__repr__>,
              '__slots__': ['_x', '_y', '_z'],
              '_x': <member '_x' of 'Point' objects>,
              '_y': <member '_y' of 'Point' objects>,
              '_z': <member '_z' of 'Point' objects>,
              'x': <property at 0x7fba69d72908>,
              'y': <property at 0x7fba69d72958>,
              'z': <property at 0x7fba69d729a8>})
```

Could we add this in ROS2 as well?

fkromer, 2018-10-29:
I don't know how to (unit) test this  easiest yet.

dirk-thomas, 2018-10-29:
The `rosidl_generator_py` package already has some unit tests which check generated messages. It should be straight forward to also check the new attribute.

fkromer, 2018-10-29:
Ah. Probably [`Various()`](https://github.com/ros2/rosidl_python/blob/master/rosidl_generator_py/test/test_interfaces.py#L190) in [`rosidl_python/rosidl_generator_py/test/test_interfaces.py`](https://github.com/ros2/rosidl_python/blob/master/rosidl_generator_py/test/test_interfaces.py) is what I should have watch out for (and missed so far).

dirk-thomas, 2019-10-24:
This should be available for a while already: ros2/rosidl_python#19.

closed 2019-10-24 by dirk-thomas