ros2/rosidl_python #40: Dependency cycle between this repository and rcl_interfaces

nuclearsandwich, 2019-04-14:
With the introduction of the `rosidl_runtime_py` package there is a dependency cycle between this repository and ros2/rcl_interfaces by way of ros2/rosidl_defaults

To make the first Dashing pre-release I have temporarily ignored the rosidl_runtime_py package in order to allow release of rosidl_defaults and rcl_interfaces but it might be worth splitting the runtime package into a separate repository to avoid this issue.

jacobperron, 2019-04-24:
Maybe I don't understand the cycle, but would removing `rosidl_runtime_py`'s dependency to `test_msgs` resolve this issue? I plan to do some refactoring as part of https://github.com/ros2/rcl_interfaces/issues/58 and it looks like removing the dependency to `test_msgs` shouldn't be too difficult.

nuclearsandwich, 2019-04-25:
> Maybe I don't understand the cycle

Bloom releases by default(1) release every package in the repository together. This means that it's possible to create a cycle between repositories. In this case:

* `rosidl_runtime_py` [depends on](https://github.com/ros2/rosidl_python/blob/191880315e01f266611dac04da3817b3ef672e81/rosidl_runtime_py/package.xml#L19) `test_msgs` from [ros2/rcl_interfaces](https://github.com/ros2/rcl_interfaces).
* The `rcl_interfaces` package in ros2/rcl_interfaces [depends on](https://github.com/ros2/rcl_interfaces/blob/99be81d6fa97c9c2f7562571eabd8f4cb782cca2/rcl_interfaces/package.xml#L18) `rosidl_default_runtime` from [ros2/rosidl_defaults](https://github.com/ros2/rosidl_defaults).
* `rosidl_default_runtime` [depends on](https://github.com/ros2/rosidl_defaults/blob/926ce4d8f460fd8d7bb8c25544c2ebe294e691c1/rosidl_default_generators/package.xml#L17) `rosidl_generator_py` from this repository.

So before being able to release this repository via bloom, we need to release rcl_interfaces, but before that, we need to release rosidl_defaults, but before that we need to release... this repository.

> would removing `rosidl_runtime_py`'s dependency to `test_msgs` resolve this issue?

I think that it would. Another option would be to split rosidl_runtime_py into its own repository that can be released independently of the repositories that depend on it as we did for the vendor typesupport packages: rosidl_typesupport_{connext,fastrtps,opensplice}.

---

(1) It's possible to ignore a package during release but this requires adding an file to the release repository, then editing it to unignore the package which is somewhat cumbersome. It also means you have to remember the cycle and how to break it for each new rosdistro you release the interdependent repositories into.

jacobperron, 2019-04-25:
Thanks for the detailed explanation :smiley:

nuclearsandwich, 2019-09-19:
@jacobperron we resolved this right?

jacobperron, 2019-09-19:
I don't think so...

https://github.com/ros2/rosidl_python/blob/81edab7636188a61b63bac857f50b217c6f2b3bf/rosidl_runtime_py/package.xml#L21

I think I thought it would get resolved during some refactoring of test interfaces, but I guess not. I'll take another look to see what can be done.

jacobperron, 2019-09-19:
Other rosidl packages get around this issue by depending on [test_interfaces_files](https://github.com/ros2/test_interface_files) and generating their own test messages (instead of depending on `test_msgs`). But `rosidl_runtime_py` is a pure Python package, so we'd have to make it a CMake package if we want it to generate test interfaces, AFAIK.

nuclearsandwich, 2019-09-19:
> But `rosidl_runtime_py` is a pure Python package, so we'd have to make it a CMake package if we want it to generate test interfaces, AFAIK.

Or create a separate ament_cmake sibling package with the required interface files.

jacobperron, 2019-09-19:
I'm proposed moving `rosidl_runtime_py` to a separate repo: https://github.com/ros2/rosidl_runtime_py/pull/1

Seems to be the easiest thing to do.

PR #88 (2019-09-19): Migrate rosidl_runtime_py to it's own repository
This helps fix a repository dependency cycle.
See #40 for details.

The package can now be found at https://github.com/ros2/rosidl_runtime_py.git

Resolves #40
closed 2019-09-19 by jacobperron