ros2/rosidl_python #67: set_message_fields given non-basic type

artivis, 2019-06-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - from source
- Version or commit hash:
  - f692e9c21cf35166603d7658e69cc710e157f5ff
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```python
from geometry_msgs.msg import Vector3, Twist
from rosidl_runtime_py import set_message_fields

def main():
    msg = Twist(/*Some Args*/)
    
    values = {}
    values['linear'] = Vector3(x=1.0, y=1.0, z=2.0)
    
    set_message_fields(msg, values)
    print(msg)
```

#### Expected behavior
Set the `msg.linear` field to `x=1.0, y=1.0, z=2.0`.
#### Actual behavior
Crash.
It works with the following tho,
```python
values['linear'] = {'x': 1.0, 'y': 1.0, 'z': 2.0}
```
#### Additional information
Output error:
```terminal
Traceback (most recent call last):
  File "build/rosidl_runtime_py/rosidl_runtime_py/set_message.py", line 43, in set_message_fields
    value = field_type(field_value)
TypeError: __init__() takes 1 positional argument but 2 were given

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "install/joy_teleop/lib/Foo/foo", line 11, in <module>
    load_entry_point('joy-teleop', 'console_scripts', 'foo')()
  File "build/joy_teleop/scripts/foo.py", line 56, in main
    set_message_fields(msg, values)
  File "build/rosidl_runtime_py/rosidl_runtime_py/set_message.py", line 46, in set_message_fields
    set_message_fields(value, field_value)
  File "build/rosidl_runtime_py/rosidl_runtime_py/set_message.py", line 36, in set_message_fields
    for field_name, field_value in values.items():
AttributeError: 'Vector3' object has no attribute 'items'
```
Temporary hack:
```terminal
diff --git rosidl_runtime_py/rosidl_runtime_py/set_message.py rosidl_runtime_py/rosidl_runtime_py/set_message.py
index 04bf3d3..dd18004 100644
--- rosidl_runtime_py/rosidl_runtime_py/set_message.py
+++ rosidl_runtime_py/rosidl_runtime_py/set_message.py
@@ -38,6 +38,8 @@ def set_message_fields(msg: Any, values: Dict[str, str]) -> None:
         field_type = type(field)
         if field_type is array.array:
             value = field_type(field.typecode, field_value)
+        elif type(field_value) is field_type:
+            value = field_value
         else:
             try:
                 value = field_type(field_value)

```

jacobperron, 2019-07-11:
Hmm.. I would have expected the assignment to work as in your example. But I guess there's no copy constructor for messages so the following line is failing:

https://github.com/ros2/rosidl_python/blob/3a1bb97183b01c2733a6c48d03cf72882ff78423/rosidl_runtime_py/rosidl_runtime_py/set_message.py#L43

@artivis I think your suggested fix will work, and then we can simplify the `else` block to

```python
else:
    value = field_type()
    set_message_fields(value, field_value)
```

Let me know if you are able to open a PR to make this change.

artivis, 2019-07-11:
@jacobperron Sure I'll do so asap.

PR #68 (2019-07-12): handle set_message_fields given some non-basic type
Fix #67
commit 2d8af1e0 (2019-07-12): handle set_message_fields given some non-basic type (#68)

Fix #67
commit 2af55d2c (2019-09-05): handle set_message_fields given some non-basic type (#68)

Fix #67
commit 777f9adc (2019-09-05): handle set_message_fields given some non-basic type (#68) (#84)

Fix #67
closed 2019-07-12 by jacobperron