ros2/rosidl_python #69: set_message_fields crash on numpy.ndarray

artivis, 2019-07-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 2d8af1e0fbb120dd9101ca620c24c59e940d1be6
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```terminal
ros2 topic pub /chatter geometry_msgs/msg/PoseWithCovarianceStamped \
> "header:
>   frame_id: ''
>   stamp:
>     nanosec: 0
>     sec: 0
> pose:
>   covariance: {}
>   pose:
>     orientation:
>       w: 0.0
>       x: 0.0
>       y: 0.0
>       z: 0.0
>     position:
>       x: 1.0
>       y: 1.0
>       z: 1.0"
```

#### Expected behavior

Publish a `PoseWithCovarianceStamped` message with the provided data.

#### Actual behavior
Crash 
```terminal
Failed to populate field: Required argument 'shape' (pos 1) not found
```

#### Additional information
This seems to come from [here](https://github.com/ros2/rosidl_python/blob/master/rosidl_runtime_py/rosidl_runtime_py/set_message.py#L47) when trying to fill the covariance which appears to be of type `numpy.ndarray`. According to [`numpy.ndarray` doc](https://docs.scipy.org/doc/numpy/reference/generated/numpy.ndarray.html), `numpy.ndarray`'s constructor take as argument the nd array shape whereas here it is provided directly with whatever the yaml tag `covariance: {}` contains.
Shouldn't `numpy.ndarray` be handled differently ?

#### Implementation considerations

jacobperron, 2019-07-16:
> Shouldn't numpy.ndarray be handled differently ?

I guess so. I would have thought that this case would cover it:

https://github.com/ros2/rosidl_python/blob/2d8af1e0fbb120dd9101ca620c24c59e940d1be6/rosidl_runtime_py/rosidl_runtime_py/set_message.py#L41

But looks like `ros2 topic pub` passes `set_message_fields` a list instead of a numpy array.

I'm not sure what the best thing to do is, since it doesn't seem possible to get the `dtype` for the numpy array based on the field type. We could infer the type of the numpy array based on the type of the element in the list, but this is not going to work for an empty list (your example (A)).

dirk-thomas, 2019-07-16:
> it doesn't seem possible to get the `dtype` for the numpy array based on the field type

That information is available in the code generator: https://github.com/ros2/rosidl_python/blob/2d8af1e0fbb120dd9101ca620c24c59e940d1be6/rosidl_generator_py/rosidl_generator_py/generate_py_impl.py#L37-L48

dirk-thomas, 2019-09-06:
Addressed by #83.

closed 2019-09-06 by dirk-thomas