ros2/rosidl_python #70: NamespacedType does not have a 'value' member

artivis, 2019-07-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 2d8af1e0fbb120dd9101ca620c24c59e940d1be6
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```python
message = import_message_from_namespaced_type(
            NamespacedType(['std_msgs', 'msg'], 'Bool'))
```

#### Expected behavior
Imports a `std_msgs/msg/Bool`
#### Actual behavior
Crash
```terminal
AttributeError: 'NamespacedType' object has no attribute 'value_type'
```
#### Additional information

Crash comes from [import_message_from_namespaced_type](https://github.com/ros2/rosidl_python/blob/master/rosidl_runtime_py/rosidl_runtime_py/import_message.py#L23).
[NamespacedType definition](https://github.com/ros2/rosidl/blob/master/rosidl_parser/rosidl_parser/definition.py#L169)

#### Implementation considerations
```terminal
diff --git rosidl_runtime_py/rosidl_runtime_py/import_message.py rosidl_runtime_py/rosidl_runtime_py/import_message.py
index ceb11ad..3952933 100644
--- rosidl_runtime_py/rosidl_runtime_py/import_message.py
+++ rosidl_runtime_py/rosidl_runtime_py/import_message.py
@@ -20,5 +20,5 @@ from rosidl_parser.definition import NamespacedType
 
 def import_message_from_namespaced_type(message_type: NamespacedType) -> Any:
     module = importlib.import_module(
-        '.'.join(message_type.value_type.namespaces))
-    return getattr(module, message_type.value_type.name)
+        '.'.join(message_type.namespaces))
+    return getattr(module, message_type.name)

```

dirk-thomas, 2019-07-19:
I would guess the annotation is just wrong. This is how it is being used:

https://github.com/ros2/rosidl_python/blob/6ec55cbcc7cc4e517eb808567ba2b742d612d3ae/rosidl_runtime_py/rosidl_runtime_py/set_message.py#L52-L53

Maybe @mikaelarguedas can comment since he wrote this in #33.

mikaelarguedas, 2019-07-20:
Yeah that's definitely a bug. I must have mixed up the types when rebasing/updating #33.

See #74 for a fix making the function behave as specified

PR #74 (2019-07-26): Make import_message_from_namespaced_type operate on NamespacedType
Fixes #70 

Instead of changing the type specification, which would either require changing the function name (breaking API) or have the name and the behavior mismatch (current state), this modifies the function to operate on `NamespacedType` instead.
This will change behavior for consumers of the function.
If this is not desired, the less disruptive alternative is to change the function specification without changing the function name or the behavior.

Added a simple test to cover the uses of the function used in this package.
closed 2019-07-26 by dirk-thomas