ros2/rosidl_python #90: char IDL type translated to Python int

jacobperron, 2019-10-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - b93c7bf09b9c6119960efc7c7dc1f16ef3330a7c
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

During #89, I noticed that interface fields of type `char` are translated to the Python `int` type.
Looking at the related [design doc](), I was expected the Python type to be `str` with length 1.
I've left a TODO in the tests and commented out tests related to `char` types since I'm not sure if this bug is in the implementation or documentation.

To reproduce, uncomment the tests related to char:

https://github.com/ros2/rosidl_python/blob/b93c7bf09b9c6119960efc7c7dc1f16ef3330a7c/rosidl_generator_py/test/test_interfaces.py#L46-L48

https://github.com/ros2/rosidl_python/blob/b93c7bf09b9c6119960efc7c7dc1f16ef3330a7c/rosidl_generator_py/test/test_interfaces.py#L63-L64

https://github.com/ros2/rosidl_python/blob/b93c7bf09b9c6119960efc7c7dc1f16ef3330a7c/rosidl_generator_py/test/test_interfaces.py#L81-L82

https://github.com/ros2/rosidl_python/blob/b93c7bf09b9c6119960efc7c7dc1f16ef3330a7c/rosidl_generator_py/test/test_interfaces.py#L291

https://github.com/ros2/rosidl_python/blob/b93c7bf09b9c6119960efc7c7dc1f16ef3330a7c/rosidl_generator_py/test/test_interfaces.py#L327-L329

https://github.com/ros2/rosidl_python/blob/b93c7bf09b9c6119960efc7c7dc1f16ef3330a7c/rosidl_generator_py/test/test_interfaces.py#L366-L367

and several other locations.

#### Expected behavior

`char` gets translated to `str` with constraint of length 1 in the generated Python code

#### Actual behavior

`char` gets translated into `int`.

#### Additional information

Since it is treated as an `int`, an array or sequence of `char` also has unexpected type `numpy.ndarray` or `array.array` respectively. I would have expected a `list` of `str`.

dirk-thomas, 2019-10-04:
Please read the very last note in the design document: http://design.ros2.org/articles/legacy_interface_definition.html

> While the mapping of `char` is unintuitive it preserves compatibility with the definition in ROS 1: “deprecated alias for `uint8`”.

jacobperron, 2019-10-04:
I see. So if we had a test that used a definition generated from an IDL interface (not a legacy ROS interface), then we should expect the type to be `str` with length 1?

dirk-thomas, 2019-10-04:
> So if we had a test that used a definition generated from an IDL interface, then we should expect the type to be str with length 1?

Yes, I would think so. (Since we probably have never exercised this we will see if there are any surprises :wink:.)

jacobperron, 2019-10-04:
I've enabled the tests for char types in legacy interfaces (#91).

closed 2019-10-07 by jacobperron