ros2/rosidl_python #106: Unable to build message packages on top of nightly archives

mikaelarguedas, 2020-04-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Focal
- Installation type:
  - nightly archive
- Version or commit hash:
  - works with archive from https://ci.ros2.org/view/packaging/job/packaging_linux/1811/
  - breaks with archive from https://ci.ros2.org/view/packaging/job/packaging_linux/1819/
- DDS implementation:
  - Connext *not* installed
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
docker pull osrf/ros2:nightly
apt update && apt install -y libssl-dev
cd && mkdir -p ws/src && cd ws
git clone https://github.com/ros2/example_interfaces src/example_interfaces
colcon build
```

#### Expected behavior

```
Starting >>> example_interfaces
Finished <<< example_interfaces [24.2s]                       

Summary: 1 package finished [24.3s]
```

#### Actual behavior

```
Starting >>> example_interfaces
--- stderr: example_interfaces                         
CMake Error at /opt/ros/foxy/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake:276 (add_dependencies):
  The dependency target "example_interfaces__rosidl_typesupport_connext_c" of
  target "example_interfaces__rosidl_typesupport_connext_c__pyext" does not
  exist.
Call Stack (most recent call first):
  /opt/ros/foxy/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/ros/foxy/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:16 (rosidl_generate_interfaces)

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< example_interfaces	[ Exited with code 1 ]

```

#### Additional information

It works with the nightly archive from  April 3rd: https://ci.ros2.org/view/packaging/job/packaging_linux/1811/
Fails with the next successfully built archive from April 11th: https://ci.ros2.org/view/packaging/job/packaging_linux/1819/

It I install connext and setup the environment it works:
```
docker pull osrf/ros2:nightly-rmw-nonfree
apt update && apt install -y libssl-dev
source /opt/rti.com/rti_connext_dds-5.3.1/resource/scripts/rtisetenv_x64Linux3gcc5.4.0.bash 
cd && mkdir -p ws/src && cd ws
git clone https://github.com/ros2/example_interfaces src/example_interfaces
colcon build
```

Issue also reported at https://answers.ros.org/question/349612/ros2-osrf-docker-container-fails-to-build/

dirk-thomas, 2020-04-17:
Please see #107 for the proposed fix. It would be great if you could verify that this makes it work for the specific case of the Docker images. I reproduced the problem locally (underlay with Connext installed, overlay with Connext purged) and was able to reproduce the same error without the patch and that the patch resolved the build.

mikaelarguedas, 2020-04-17:
Dropping this in place at `/opt/ros/foxy/share/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake` fixes it for the nightly docker image :+1:

PR #107 (2020-04-17): skip inoperable typesupport implementations
Fixes #106.

Before #102 the available RMW packages where enumerated and `rmw_connext_cpp` returned `not-found` if Connext wasn't available: https://github.com/ros2/rmw_connext/blob/ae1bc361b1ee77be826b37bc3d0b2a0d6caa34ec/rmw_connext_cpp/rmw_connext_cpp-extras.cmake#L21
After #102 RMW impl. aren't considered anymore but only the type support packages.

This patch checks for the actual existence of the typesupport specific target which is about to be used instead of any secondary state (package existence, index registration, extensions, etc.).
closed 2020-04-17 by dirk-thomas