ros2/rosidl_python #114: linking issue ubuntu 20.04 foxy

flynneva, 2020-05-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - from source
- Version or commit hash:
  - 40f4d053fa846e56a9583800466bd4cae33ce2d7
- DDS implementation:
  - the default (...I think Fast-RTPS)
- Client library (if applicable):
  - NA

#### Steps to reproduce issue
- follow the foxy install from source directions for ubuntu 20.04 

```
colcon build
```

#### Expected behavior
- compiles fine

#### Actual behavior
<details><summary>colcon build output</summary>
<p>

```bash
--- stderr: rosidl_generator_py                                                           
/usr/bin/ld: /usr/local/lib/libpython3.8.a(abstract.o): relocation R_X86_64_PC32 against symbol `PyObject_SetItem' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rosidl_generator_py_custom__python.dir/build.make:256: rosidl_generator_py/rosidl_generator_py/librosidl_generator_py_custom__python.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:329: CMakeFiles/rosidl_generator_py_custom__python.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rosidl_generator_py	[ Exited with code 2 ]
```
</p>
</details>

#### Additional information
- i am getting a similar issue for qt_gui_cpp as well, no solution there yet either

flynneva, 2020-05-17:
I think it might be because of [this issue](https://root-forum.cern.ch/t/recompile-with-fpic-or-libpython3-6m-a-needed-by-lib-libpyroot-so/30206). going to wipe clean my computer and reinstall the 20.04 OS. it mightve been because I had multiple versions of python installed before upgrading - including python 3.8.

hidmic, 2020-05-21:
@flynneva let us know what you find and if we should close this issue.

flynneva, 2020-05-21:
yeah i think my issue was that I had multiple python versions installed when I upgraded to 20.04. doing a clean install of 20.04 and reinstalling everything ended up fixing the problem.

closed 2020-05-21 by flynneva