ros2/rosidl_python #135: Backport array optimization to Galactic and Foxy

sloretz, 2021-06-11:
This is an issue to backport #129 to Galactic and Foxy. I think it would be a good idea to have it sit in Rolling a little bit to make sure it works since it changes something so low level.

pulkitgoyal1506, 2021-10-05:
@sloretz Can we merge this patch on foxy now?
I am facing this issue on `ros2::foxy`, it would be great if this can be merged.

sloretz, 2021-10-09:
Released to Galactic in `0.11.1` https://github.com/ros/rosdistro/pull/30955

hidmic, 2021-12-23:
@sloretz have we backported this to Foxy already?

sloretz, 2021-12-23:
Yeah it's been backported. It might need a foxy release

jacobperron, 2022-07-22:
This has also been released.

closed 2022-07-22 by jacobperron

travipross, 2022-08-24:
> This has also been released into Foxy.

As in, fetching the latest container from Dockerhub for `ros2:foxy` should include this fix? 

I see this issue is listed as "Needs release" in the [Foxy Release Patch 8 ](https://github.com/orgs/ros2/projects/36) project, so I wasn't sure whether to expect this to exist in the official Docker image.

jacobperron, 2022-09-06:
@travipross The patch was backported to Foxy in https://github.com/ros2/rosidl_python/pull/146 and released in 0.9.5. It looks like 0.9.5 has also been sync'd to the main release repository: http://repo.ros2.org/status_page/ros_foxy_default.html?q=rosidl_python
So, the binary package should be available in Ubuntu. I'm not involved in maintaining the Docker image for Foxy, so I'm not sure what is required for updating it.

jacobperron, 2022-09-06:
And, the Foxy Patch release board is lagging behind a bit since I have to update it manually.