ros2/rosidl_python #141: Build error when using ament_python_install_package() and generated interfaces

jtbandes, 2021-08-19:
Adapted from a comment on #131:

This change prevents a package from building if it both generates interfaces and installs a Python package. For example, I have a package that is not longer building since it does something like this:

```cmake
rosidl_generate_interfaces(${PROJECT_NAME}_msgs
  ...
)

ament_python_install_package(${PROJECT_NAME})
```

I see an error like:

```
CMake Error at /home/jacob/ros2-linux/share/ament_cmake_python/cmake/ament_python_install_package.cmake:108 (add_custom_target):
  add_custom_target cannot create target
  "ament_cmake_python_symlink_my_package" because another target with the
  same name already exists.  The existing target is a custom target created
  in source directory
  ...
```

_Originally posted by @jacobperron in https://github.com/ros2/rosidl_python/issues/131#issuecomment-856221882_

jtbandes, 2021-08-19:
I'm a CMake noob, but would it be possible to simply pass a different, unique target name on this line instead of just `${PROJECT_NAME}`?

https://github.com/ros2/rosidl_python/blob/191a1435be9df09a1c2945e8558300c566e4219d/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake#L123-L125

jtbandes, 2021-09-29:
This issue is blocking packaging of rosbridge_suite for rolling (https://github.com/RobotWebTools/rosbridge_suite/issues/661). @clalancette @jacobperron Any ideas how the issue can be resolved? Has any progress been made on it yet?

clalancette, 2021-09-30:
I would say the easiest workaround is to split the Python code and the message generation into two separate packages.  You generally want to do that anyway; a downstream client may want to communicate using your custom messages, but may not want/need the other code that comes along with it.

jacobperron, 2021-10-05:
@jtbandes This dropped off my radar, sorry for the late reply.

> One thing I don't understand is why this error appears for rosapi, but not rosbridge_library.

I would expect the error to appear for both packages. I'm guessing it has something to do with the package.xml; there's a discrepancy between the package.xml of rosbridge_library and rosapi. Notably, rosbridge_library does not declare itself as an interface package (ie. with `<member_of_group>rosidl_interface_packages</member_of_group>`). This is just a guess.

> I'm a CMake noob, but would it be possible to simply pass a different, unique target name on this line instead of just ${PROJECT_NAME}?

I think this is doable. Though, since https://github.com/ros2/rosidl_python/pull/131, I think the main issue is now we risk install collisions between `rosidl_generate_interfaces` and your custom package. @hidmic is this accurate?

> Any ideas how the issue can be resolved? Has any progress been made on it yet?

I'm not really sure how to proceed. Maybe we provide an option for users install multiple Python packages with the same name? We could try to detect actual install collisions and error in that case. I don't think anyone has looked into this or proposed other options. I worked around the issue originally by creating a separate package for my interfaces :shrug: 

>  is there anything we could have done in either the rosbridge_suite repo or this repo to catch this kind of failure sooner, and not have to wait for another Rolling release to incorporate a fix into rosbridge_suite

I think this is what Rolling is meant for: allow maintainers to build/test against the latest changes and fix things in preparation for the next ROS release. You could setup a CI job for your repo that builds against Rolling. I suggest using our Debian _testing_ repository to avoid having to wait for the next Rolling sync ([instructions here](https://docs.ros.org/en/rolling/Installation/Testing.html#debian-testing-repository)).

jacobperron, 2021-10-05:
Another workaround to consider is renaming your Python package such that it does not match the package name.

jtbandes, 2021-10-12:
Ok, that makes sense. It seems like the easiest thing to do is release the `rosbridge_test_msgs` package. Do you have any concerns with that approach?

hidmic, 2021-10-21:
> I think this is doable. Though, since #131, I think the main issue is now we risk install collisions between rosidl_generate_interfaces and your custom package. @hidmic is this accurate?

Yes! This fell off my radar completely. 

---

To solve this, perhaps we can rehash `ament_cmake_python` CMake macros to be a bit more idiomatic. As in, have a macro that  add a module (i.e. a collection of files and/or subdirectories) and then have a macro that installs a list of modules as a package. `ament_python_install_package`  is already reconstructing the source package in the build tree for `setuptools` to build the egg, so it's not that far off. Thoughts @jacobperron @clalancette ?

hidmic, 2021-10-21:
> this still puts the responsibility on the user to avoid install collisions if they are installing interfaces and custom Python code to the same module.

To the extent that user provided and generated `.py` sources must not overwrite each other, yes. We can detect duplication at configure time though and fail the build early.

RFRIEDM-Trimble, 2022-07-14:
Can we add a warning on the [tutorial](https://docs.ros.org/en/humble/How-To-Guides/Ament-CMake-Python-Documentation.html) that this isn't supported?

clalancette, 2022-07-14:
> Can we add a warning on the [tutorial](https://docs.ros.org/en/humble/How-To-Guides/Ament-CMake-Python-Documentation.html) that this isn't supported?

Sure, please feel free to open a PR there and we'd be happy to review.

knorth55, 2022-10-26:
I have this issue in `sound_play`.

hmmm, it seems we must separate python package and message package.
i have a question, but message package must be a separate package?
I know it is a good practice, but I think it is just a tradition or a custom habit.
If it is ruled, it should be written in REP, I think.

clalancette, 2022-10-26:
> i have a question, message package must be a separate package?

Currently, yes, due to this bug.

> However, if it is not, this issue should be fixed.

Yes, this bug should be fixed.  Someone has to find the time to do it.

knorth55, 2022-10-26:
I made a hotfix #187.
In this PR, ament_python_install_package is called when the python package is not installed.

brta-jc, 2023-10-15:
Any progress on this? This is still a huge issue. Propose we merge @knorth55 hotfix 187 unless work is being done to fix the bug.

peterdavidfagan, 2023-10-24:
+1 would be helpful to have this fixed.

ozarrai, 2024-01-05:
+1 would be great if this bug is fixed.

clalancette, 2024-03-08:
So we discussed this bug a bit.

The most `ament_cmake` way to do this is to change up the way that `ament_python_install_package` works.  For most things that are part of `ament_cmake`, the way that they work is that the call (to e.g. [ament_export_libraries](https://github.com/ament/ament_cmake/tree/rolling/ament_cmake_export_libraries)) doesn't do much work, and instead sets up a bunch of cmake variables, along with a hook.  When `ament_package` is eventually called, it calls all of the hooks, and at that point the hook does the work based on the cmake variables.  This is done this way so that `ament_package` has a full view of the environment, and avoids problems exactly like this.

For whatever reason, `ament_python_install_package` does work like this at the moment.  We should change `ament_python_install_package` to work in the more `ament_cmake` way, which will resolve this bug.

What I'm going to do here is to close this issue (and related PR) in favor of https://github.com/ament/ament_cmake/issues/514 (partially because GitHub doesn't allow us to transfer issues between organizations).  Then we can continue the conversation over there (I'll also replicate this comment there).

closed 2024-03-08 by clalancette