ros2/rosidl_python #143: extra parenthesis in generated file

flynneva, 2021-09-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - galactic
- DDS implementation:
  - NA
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
cd colcon_ws/src
git clone https://github.com/ApexAI/reference-system-autoware.git

cd colcon_ws
colcon build
colcon test
```

the logs for `reference_interfaces` should show the error during the flake8 tests.

#### Expected behavior

generated files do not throw lint errors and can be run without error

#### Actual behavior

Generated file has this error when passed into the python interpreter:

```
build/reference_interfaces/rosidl_generator_py/reference_interfaces/msg$ python3 _transmission_stats.py 
  File "_transmission_stats.py", line 167
    all(val >= 0 and val) < 256 for val in value)), \
                                                 ^
SyntaxError: unmatched ')'
```

it looks like an extra parenthesis was added to the file.

#### Additional information

this is [the offending idl file](https://github.com/ApexAI/reference-system-autoware/blob/feature/unit_tests/reference_interfaces/msg/TransmissionStats.idl)

flynneva, 2021-09-13:
looks like its coming from [this line](https://github.com/ros2/rosidl_python/blob/7638b52d325a0efd2250786300ccb40d83a4a291/rosidl_generator_py/resource/_msg.py.em#L498) which _I think_ does in fact have an extra parenthesis

sloretz, 2021-09-13:
CI (repos: `https://gist.githubusercontent.com/sloretz/5907cc90bb01ccb11c780949f11454a0/raw/75a1a69d7acb30dc6549d66c79801aed5d6cc09e/ros2.repos` build: `--packages-above-and-dependencies rosidl_generator_py` test: `--packages-above rosidl_generator_py`)
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

flynneva, 2021-09-16:
if anyone else ends up here before the fix is released, just add this to the bottom of your package to automatically fix, modifying as needed. the key is the `TARGET` name being your package name + `__python` to wait for `rosidl_generator_py` to do its thing:

```
add_custom_command(
  TARGET ${PROJECT_NAME}__python
  POST_BUILD
  COMMAND sed -i "s/all(val >= 0 and val) < 256/all(ord(val) >= 0 and ord(val) < 256/" ${GENERATED_FILE}
  COMMENT "Check generated IDL files for extra parenthesis..."
  VERBATIM)
  ```

closed 2022-02-10 by clalancette