ros2/rosidl_python #150: 'property' object is not callable

fujitatomoya, 2021-12-14:
## Bug report

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/4a36f319afcc208deae3c373493ef6d9feda1324
- DDS implementation:
  - anything (cyclonedds, Fast-DDS, ConnextDDS)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

the following colcon workspace includes reproducible package.

```bash
### how to build
cd <your colcon workspace>/src/ros2
git clone git@github.com:fujitatomoya/ros2_test_prover.git
cd <your colcon workspace>
colcon build --symlink-install --packages-select prover_rclcpp prover_rclpy prover_interfaces
source install/local_setup.bash

### terminal-1
ros2 run prover_rclcpp ros2cli_679
...

### terminal-2
ros2 topic hz /property
Traceback (most recent call last):
  File "/root/ros2_ws/master_ws/install/ros2cli/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli', 'console_scripts', 'ros2')())
  File "/root/ros2_ws/master_ws/build/ros2cli/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/verb/hz.py", line 87, in main
    return main(args)
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/verb/hz.py", line 102, in main
    _rostopic_hz(node.node, topic, window_size=args.window_size, filter_expr=filter_expr,
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/verb/hz.py", line 256, in _rostopic_hz
    msg_class = get_msg_class(
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/api/__init__.py", line 98, in get_msg_class
    msg_class = _get_msg_class(node, topic, include_hidden_topics)
  File "/root/ros2_ws/master_ws/build/ros2topic/ros2topic/api/__init__.py", line 144, in _get_msg_class
    return get_message(message_type)
  File "/root/ros2_ws/master_ws/build/rosidl_runtime_py/rosidl_runtime_py/utilities.py", line 28, in get_message
    interface = import_message_from_namespaced_type(get_message_namespaced_type(identifier))
  File "/root/ros2_ws/master_ws/build/rosidl_runtime_py/rosidl_runtime_py/import_message.py", line 30, in import_message_from_namespaced_type
    module = importlib.import_module(
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/root/ros2_ws/master_ws/install/prover_interfaces/lib/python3.8/site-packages/prover_interfaces/msg/__init__.py", line 1, in <module>
    from prover_interfaces.msg._property import Property  # noqa: F401
  File "/root/ros2_ws/master_ws/install/prover_interfaces/lib/python3.8/site-packages/prover_interfaces/msg/_property.py", line 52, in <module>
    class Property(metaclass=Metaclass_Property):
  File "/root/ros2_ws/master_ws/install/prover_interfaces/lib/python3.8/site-packages/prover_interfaces/msg/_property.py", line 131, in Property
    def anything(self):
TypeError: 'property' object is not callable
```

#### Expected behavior

`ros2 topic` command issued successfully.

#### Actual behavior

failed with `'property' object is not callable`

#### Additional information

This problem only happens if message type is defined after `property` message type.

https://github.com/fujitatomoya/ros2_test_prover/blob/master/prover_interfaces/msg/Property.msg

```
string property
string anything
```

clalancette, 2021-12-14:
> btw, can you move this issue to https://github.com/ros2/rosidl_python/issues?

Done now.

iuhilnehc-ynos, 2021-12-15:
What about using `@builtins.property` explicitly instead of `@property`?

closed 2021-12-17 by hidmic