ros2/rosidl_python #154: Error: Undefined symbols _PyIntl_bind_textdomain_codeset in libpython3.8.a (rosidl_generator_py)

agorskih, 2022-02-21:
## Bug report

**Required Info:**

- Operating System:
  - macOS Monterey 12.0.1 (21A559)
- Installation type:
  - from source
- Version or commit hash:
  - repos.file:
                ros2/rosidl_python:
                type: git
                url: https://github.com/ros2/rosidl_python.git
                version: galactic
- DDS implementation:
  - rmw_cyclonedds
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```
git clone https://github.com/ros2/ros2.git ros2_galactic
cd ros2_galactic
mkdir ./src
vcs import src < ros2.repos
colcon build --cmake-clean-cache --packages-select rosidl_generator_py
```

#### Expected behavior
Starting >>> rosidl_generator_py
Finished <<< rosidl_generator_py [9.96s]                   

Summary: 1 package finished [12.1s]

#### Actual behavior
```
Starting >>> rosidl_generator_py
--- stderr: rosidl_generator_py                              
Undefined symbols for architecture x86_64:
  "_libintl_bind_textdomain_codeset", referenced from:
      _PyIntl_bind_textdomain_codeset in libpython3.8.a(_localemodule.o)
  "_libintl_bindtextdomain", referenced from:
      _PyIntl_bindtextdomain in libpython3.8.a(_localemodule.o)
  "_libintl_dcgettext", referenced from:
      _PyIntl_dcgettext in libpython3.8.a(_localemodule.o)
  "_libintl_dgettext", referenced from:
      _PyIntl_dgettext in libpython3.8.a(_localemodule.o)
  "_libintl_gettext", referenced from:
      _PyIntl_gettext in libpython3.8.a(_localemodule.o)
  "_libintl_setlocale", referenced from:
      _PyLocale_setlocale in libpython3.8.a(_localemodule.o)
      _PyLocale_localeconv in libpython3.8.a(_localemodule.o)
  "_libintl_textdomain", referenced from:
      _PyIntl_textdomain in libpython3.8.a(_localemodule.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [rosidl_generator_py/rosidl_generator_py/librosidl_generator_py_custom__python.dylib] Error 1
make[1]: *** [CMakeFiles/rosidl_generator_py_custom__python.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< rosidl_generator_py [9.34s, exited with code 2]

Summary: 0 packages finished [10.9s]
  1 package failed: rosidl_generator_py
  1 package had stderr output: rosidl_generator_py
```

#### Additional information
1. build/rosidl_generator_py/CMakeFiles/rosidl_generator_py_custom__rosidl_generator_c.dir/link.txt contains -L/usr/local/opt/gettext/lib as it should.
2. /usr/local/opt/gettext/lib has both libintl.dylib and libintl.a. nm -gU shows that symbols are there.
3. python 3.8.12 interpreter used also can access gettext library with no problem
4. commenting following code in src/ros2/rosidl_python/rosidl_generator_py/CMakeLists.txt allows compilation to proceed:
```
rosidl_generator_py_extras(
    "${CMAKE_CURRENT_SOURCE_DIR}/bin/rosidl_generator_py"
    "${CMAKE_CURRENT_SOURCE_DIR}/rosidl_generator_py/__init__.py;${CMAKE_CURRENT_SOURCE_DIR}/rosidl_generator_py/generate_py_impl.py"
    "${CMAKE_CURRENT_SOURCE_DIR}/resource"
  )
```
5. lines that result in an error is the following:
```
rosidl_generate_interfaces(${PROJECT_NAME}_custom
      ${test_interface_files_MSG_FILES}
      # Cases not covered by test_interface_files
      msg/StringArrays.msg
      ADD_LINTER_TESTS
      SKIP_INSTALL
    )
```

How can I debug this further? Any suggestions why linker can be missing gettext related libraries?

sloretz, 2022-02-22:
> How can I debug this further? Any suggestions why linker can be missing gettext related libraries?

I would recommend trying a build with `CMAKE_VERBOSE_MAKEFILE` to get the compile commands, and then debug by looking at the command that is failing.

```
colcon build --cmake-clean-cache -DCMAKE_VERBOSE_MAKEFILE=ON --packages-select rosidl_generator_py
```

If the error is in the call to `rosidl_generate_interfaces`, it would seem likely that the fix would be making the target created in the extension `rosidl_generator_py` provides call `target_link_libraries()` on some python library to give the linker access to the symbol. So, the fix will probably go into this file:

https://github.com/ros2/rosidl_python/blob/master/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake

Maybe `PythonExtra_LIBRARIES` isn't enough?

https://github.com/ros2/rosidl_python/blob/4acc8fc6098f9fd1f0172c42d9422d35527b411c/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake#L175-L178

That variable comes from `python_cmake_module`: https://github.com/ros2/python_cmake_module/blob/master/cmake/Modules/FindPythonExtra.cmake

agorskih, 2022-02-23:
Thank you for your help. After adding following code to rosidl_generator_py_generate_interfaces.cmake (line 189) build succeed: 
```
target_link_libraries(
  ${_target_name_lib}
  "/usr/local/opt/gettext/lib/libintl.8.dylib"
)
```
I will test it on clean MacOS instance to make sure this fixes installation issue on new MacOS version.
What is a better place to put libintl.8.dylib dependency for MacOS version?
BTW, Is there a CI setup that tests MacOS installation of ROS2?

clalancette, 2022-02-23:
> BTW, Is there a CI setup that tests MacOS installation of ROS2?

No, macOS is currently a Tier 3, community supported distribution (as listed in https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027).  That said, we are happy to review patches that fix problems on macOS.

closed 2022-02-23 by agorskih