ros2/rosidl_python #168: "cannot import name 'generate_py' from 'rosidl_generator_py'" when going through workspace tutorial

aliddell, 2022-06-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - Binaries
- Version or commit hash:
  - (I'm guessing this is the relevant package:
  - Package: `ros-rolling-rosidl-generator-py`
  - Version: `0.15.0-1jammy.20220521.000012`

- DDS implementation:
  - (Guessing) Fast-RTPS.
- Client library (if applicable):
  - Not sure if applicable.

#### Steps to reproduce issue

Run through [this tutorial](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html). I'm observing this in rolling and humble.

```shell
$ source /opt/ros/rolling/setup.bash
$ mkdir -p ~/dev_ws/src && cd ~/dev_ws/src
$ git clone https://github.com/ros/ros_tutorials.git -b rolling-devel
$ cd ~/dev_ws && rosdep install -i --from-path src --rosdistro rolling -y
$ colcon build
```

#### Expected behavior

Turtlesim builds.

#### Actual behavior

Receive the following output:

```
[0.270s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'turtlesim' is in: /opt/ros/rolling
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding turtlesim

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> turtlesim
--- stderr: turtlesim                             
Traceback (most recent call last):
  File "/opt/ros/rolling/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ImportError: cannot import name 'generate_py' from 'rosidl_generator_py' (/opt/ros/rolling/local/lib/python3.10/dist-packages/rosidl_generator_py/__init__.py)
gmake[2]: *** [turtlesim__py/CMakeFiles/turtlesim__py.dir/build.make:121: rosidl_generator_py/turtlesim/_turtlesim_s.ep.rosidl_typesupport_fastrtps_c.c] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:706: turtlesim__py/CMakeFiles/turtlesim__py.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< turtlesim [0.33s, exited with code 2]

Summary: 0 packages finished [0.45s]
  1 package failed: turtlesim
  1 package had stderr output: turtlesim
```

#### Additional information

If this is the wrong place to report this error, do point me in the right direction.

clalancette, 2022-06-02:
Hm, that's odd.  I can't reproduce the problem here; the turtlesim package builds fine for me.

Can you go into a fresh terminal (where you haven't sourced anything ROS related), and then do:

```
source /opt/ros/rolling/setup.bash
python3 -c 'from rosidl_generator_py import generate_py'
```

I'm assuming that is going to fail as well.  If it does, I think it bears looking into the contents of `/opt/ros/rolling/local/lib/python3.10/dist-packages/rosidl_generator_py/generate_py_impl.py` to ensure that the `generate_py` method exists in that file, as that is where it should be coming from.

clalancette, 2022-06-02:
> `rosidl_generator_py_root` actually points to `/opt/ros/rolling/lib`, so that `rosidl_generator_py_module` is looking for `/opt/ros/rolling/lib/rosidl_generator_py/__init__.py`, which doesn't exist (because the only file in that directory is the script above). So I'm not sure if this is an issue in packaging or if there were some symlinks that should have been created and weren't, or what.

Yeah, that doesn't exist, but it shouldn't need to.  That fallback should only ever be used on Windows.  The real thing that is failing is `from rosidl_generator_py import generate_py`, which should be succeeding.

I just tried a fresh installation of Jammy and Rolling following the directions at https://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html , and I'm able to successfully import:

```
$ source /opt/ros/rolling/setup.bash
$ python3 -c 'from rosidl_generator_py import generate_py'
$
```

What were the exact steps you used to install Rolling?

Separately, are you maybe using a python virtual environment or something like that?

closed 2022-06-02 by aliddell

minkowski-jetson, 2022-09-07:
I'm having this issue on Ubuntu 22.04. I believe humble packages are supposed to be built with python3.6. If I use the system python which is 3.10, builds fail for various incompatibility reasons with python3.10. If I use pyenv to build with python3.6 I get the error that this issue relates to. 

It's a conundrum because ros humble seems to require system python which is 3.10, and workarounds that use the correct python version (anaconda, pyenv) cause this package to fail.

IamPhytan, 2022-10-01:
Hello, I have the same problem as @aliddell, without any Python/Conda environment that could have an effect on the import

It works when I import it simply with `python3`, but it doesn't work when I `colcon build`.

```sh
user@hostname:~/tutorial_ws$ python3 -c 'from rosidl_generator_py import generate_py'
user@hostname:~/tutorial_ws$ colcon build --packages-select tutorial_interfaces 
Starting >>> tutorial_interfaces
--- stderr: tutorial_interfaces                             
Traceback (most recent call last):
  File "/opt/ros/humble/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ImportError: cannot import name 'generate_py'
gmake[2]: *** [tutorial_interfaces__py/CMakeFiles/tutorial_interfaces__py.dir/build.make:157: rosidl_generator_py/tutorial_interfaces/_tutorial_interfaces_s.ep.rosidl_typesupport_fastrtps_c.c] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:590: tutorial_interfaces__py/CMakeFiles/tutorial_interfaces__py.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< tutorial_interfaces [1.89s, exited with code 2]

Summary: 0 packages finished [2.07s]
  1 package failed: tutorial_interfaces
  1 package had stderr output: tutorial_interfaces
```

My current workaround is to comment out this line in the `CMakeLists.txt` : 

```cmake
# find_package(rosidl_default_generators REQUIRED)
```

mrayh001, 2023-08-17:
Hi, I am also facing the same issue with ros2 humble and ubuntu 22.04

when i do python3 -c 'from rosidl_generator_py import generate_py'
I can import. 
I can also import with python
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from rosidl_generator_py import generate_py
>>> 

But when I do colcon build it says 
Traceback (most recent call last):
  File "/opt/ros/humble/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ImportError: cannot import name 'generate_py'
gmake[2]: *** [spot_msgs__py/CMakeFiles/spot_msgs__py.dir/build.make:1497: rosidl_generator_py/spot_msgs/_spot_msgs_s.ep.rosidl_typesupport_fastrtps_c.c] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:590: spot_msgs__py/CMakeFiles/spot_msgs__py.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< spot_msgs [43.2s, exited with code 2]

overtog, 2024-11-13:
I have the same issue with different version of python3 installed on linux
It works fine with python 3.10 but fail with predecessors
Just to remove other version of python3 helped me

ShenoyVaradaraya, 2025-03-09:
I fixed it by adding `/usr/bin/python3/dist-packages` in myzshrc/bashrc.