ros2/rosidl_python #174: Failed to build custom message

PenguineZhang, 2022-07-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS version:
  - Humble
- Installation type:
  - Binary
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
- create custom ros2 messages (msg file only contains one uint8 type)
- build the custom ros2 message with `colcon build --packages-select tello_msgs`

#### Expected behavior
Build success
#### Actual behavior
$ colcon build --packages-select tello_msgs
Starting >>> tello_msgs
--- stderr: tello_msgs                             
Traceback (most recent call last):
  File "/opt/ros/humble/share/rosidl_generator_py/cmake/../../../lib/rosidl_generator_py/rosidl_generator_py", line 8, in <module>
    from rosidl_generator_py import generate_py
ImportError: cannot import name 'generate_py' from 'rosidl_generator_py' (/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_generator_py/__init__.py)
gmake[2]: *** [tello_msgs__py/CMakeFiles/tello_msgs__py.dir/build.make:130: rosidl_generator_py/tello_msgs/_tello_msgs_s.ep.rosidl_typesupport_fastrtps_c.c] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:590: tello_msgs__py/CMakeFiles/tello_msgs__py.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< tello_msgs [0.48s, exited with code 2]

# Addition information
CmakeLists.txt:

cmake_minimum_required(VERSION 3.8)
project(tello_msgs)

if (NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif ()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rosidl_default_generators REQUIRED)

rosidl_generate_interfaces(
  ${PROJECT_NAME}
  "msg/TelloMsg.msg"
)

ament_package()
-------------------------------------------------------------------------------------------------------
package.xml:

<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>tello_msgs</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="todo@todo.todo">todo</maintainer>
  <license>TODO: License declaration</license>

  <buildtool_depend>ament_cmake</buildtool_depend>

  <build_depend>rosidl_default_generators</build_depend>

  <exec_depend>rosidl_default_runtime</exec_depend>

  <member_of_group>rosidl_interface_packages</member_of_group>
  
  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>

sloretz, 2022-07-25:
I'm unable to reproduce. What version of `rosidl_generator_py` is installed? Here's what's installed on my machine.

```console
$ apt show ros-humble-rosidl-generator-py
Package: ros-humble-rosidl-generator-py
Version: 0.14.2-2jammy.20220520.005932
Status: install ok installed
Priority: optional
Section: misc
Maintainer: Michel Hidalgo <michel@ekumenlabs.com>
Installed-Size: 197 kB
Depends: python3-numpy, ros-humble-ament-cmake, ros-humble-ament-index-python, ros-humble-python-cmake-module, ros-humble-rmw, ros-humble-rosidl-cli, ros-humble-rosidl-cmake, ros-humble-rosidl-generator-c, ros-humble-rosidl-parser, ros-humble-rosidl-runtime-c, ros-humble-rosidl-typesupport-c, ros-humble-rosidl-typesupport-interface, ros-humble-rpyutils, ros-humble-ros-workspace
Download-Size: unknown
APT-Manual-Installed: no
APT-Sources: /var/lib/dpkg/status
Description: Generate the ROS interfaces in Python.
```

Here's the package I built: [tello_msgs.tar.gz](https://github.com/ros2/rosidl_python/files/9185455/tello_msgs.tar.gz)

>   <build_depend>rosidl_default_generators</build_depend>

One nitpick, I think this should be a `<buildtool_depend>`

PenguineZhang, 2022-07-25:
I ended up reinstalling Ubuntu and now the problem is gone.

closed 2022-07-25 by PenguineZhang