ros2/rosidl_python #186: rosidl_python installs two almost identical versions of lib<PACKAGE_NAME>__rosidl_generator_py.so

DasRoteSkelett, 2022-10-05:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 22.04, yocto honister
- Installation type:
  binaries, from source
- Version or commit hash:
  - humble, rolling, 20bcc4ca118145a8a2914fa20aaef20a8e3ce72d
- DDS implementation:
  - default

#### Steps to reproduce issue
If you look for files that end with `__rosidl_generator_py.so` you will find two versions of the same library. The only difference seems to be the RUNPATH that is set to `/tmp/binarydeb/ros-humble-builtin-interfaces-1.2.0/.obj-x86_64-linux-gnu:/opt/ros/humble/lib:` (ymmv here)

``` 
cd /opt/ros/humble/
ROSIDL_FROM_LIB=./lib/libbuiltin_interfaces__rosidl_generator_py.so 
ROSIDL_FROM_DIST_PACKAGES=./local/lib/python3.10/dist-packages/builtin_interfaces/libbuiltin_interfaces__rosidl_generator_py.so
ls -l ${ROSIDL_FROM_LIB} ${ROSIDL_FROM_DIST_PACKAGES}
echo "Comparing the contents:"
diff <(nm -D  ${ROSIDL_FROM_LIB}) <(nm -D ${ROSIDL_FROM_DIST_PACKAGES})
md5sum ${ROSIDL_FROM_LIB} ${ROSIDL_FROM_DIST_PACKAGES}
diff <(readelf -a ${ROSIDL_FROM_LIB}) <(readelf -a ${ROSIDL_FROM_DIST_PACKAGES})
```

#### Expected behavior

Only one version gets installed (e.g. foxy)

#### Actual behavior

Two almost identical versions get installed

#### Additional information

This popped up while trying to get humble to work with [meta-ros](https://github.com/ros/meta-ros) where this is a QA issue 
````
ERROR: builtin-interfaces-1.2.0-2-r0 do_package: builtin-interfaces: Multiple shlib providers for libbuiltin_interfaces__rosidl_generator_py.so: builtin-interfaces, builtin-interfaces (used by files: /yocto/sf-build/BUILD-ros2-humble-honister/work/core2-64-oe-linux/builtin-interfaces/1.2.0-2-r0/packages-split/builtin-interfaces/usr/lib/python3.9/site-packages/builtin_interfaces/builtin_interfaces_s__rosidl_typesupport_fastrtps_c.cpython-39-x86_64-linux-gnu.so)
```
I will have a look into this, but since I am not very familiar with this, I do not think I get this done quick or without side effects.

RethinkMatze, 2022-12-16:
*Any* progress or statement on this and / or on the pull request?

cottsay, 2023-04-27:
I haven't verified this, but I believe that this is where the library gets installed to `lib`: https://github.com/ros2/rosidl_python/blob/52eaf342deea5fda49d9589deaad6fbb481a6221/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake#L302-L306

...and I believe that it gets installed into `site-packages` because the library output directory is getting overridden, and I'm not sure it should be: https://github.com/ros2/rosidl_python/blob/52eaf342deea5fda49d9589deaad6fbb481a6221/rosidl_generator_py/cmake/rosidl_generator_py_generate_interfaces.cmake#L156-L159

So figuring out why the output directory is getting overridden is the path forward.

RethinkMatze, 2023-04-28:
I made a pull request that fixes this:

https://github.com/ament/ament_cmake/pull/413

So far, nobody cared to look at it.

clalancette, 2023-04-28:
> [ament/ament_cmake#413](https://github.com/ament/ament_cmake/pull/413)

I don't think that is the right solution.  In particular, it seems to fix the symptom (an extra library), rather than the cause (the extra library being created in the first place).  I think we probably need to investigate deeper into `rosidl_python` itself and just not generate that extra library to begin with.

RethinkMatze, 2023-04-28:
> I don't think that is the right solution. In particular, it seems to fix the symptom (an extra library), rather than the cause (the extra library being created in the first place). I think we [...]

Well, I know I have spend some time back then, but since it is some time ago, I do not remember exactly the reasoning for this solution. Interestingly it is some kind of regression, as foxy did not have those extra libraries. Therefore, if you have other suggestions or concrete files / functions to look at I will be happy to improve the PR to a state that it can considered to be merged.

brta-jc, 2023-10-15:
This is still an issue. We need to split packages into '_msgs' packages, which for some packages is really unnecessary and cumbersome. Any updates on the PR's? Can we have the pre foxy behavior back please?

closed 2023-11-09 by clalancette