ros2/rosidl_python #213: Custom message don't compile with -Wl,--no-undefined

Rayman, 2024-06-18:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 24.04
- Installation type: binaries
- Version or commit hash: 4.6.2-1noble.20240513.234614
- DDS implementation: default
- Client library (if applicable): N/A

#### Steps to reproduce issue

```
docker run -it ros:jazzy
cd && mkdir src && cd src
git clone https://github.com/Rayman/tutorial_interfaces.git
cd
colcon build
```

#### Expected behavior
Custom messages will compile with `-Wl,--no-undefined` linker flag.

#### Actual behavior
```
Starting >>> tutorial_interfaces
--- stderr: tutorial_interfaces                              
/usr/bin/ld: CMakeFiles/tutorial_interfaces_s__rosidl_typesupport_introspection_c.dir/rosidl_generator_py/tutorial_interfaces/_tutorial_interfaces_s.ep.rosidl_typesupport_introspection_c.c.o: in function `Py_XDECREF':
_tutorial_interfaces_s.ep.rosidl_typesupport_introspection_c.c:(.text+0x5a): undefined reference to `_Py_Dealloc'
/usr/bin/ld: CMakeFiles/tutorial_interfaces_s__rosidl_typesupport_introspection_c.dir/rosidl_generator_py/tutorial_interfaces/_tutorial_interfaces_s.ep.rosidl_typesupport_introspection_c.c.o: in function `_register_msg_type__msg__num':
_tutorial_interfaces_s.ep.rosidl_typesupport_introspection_c.c:(.text+0xc7): undefined reference to `PyCapsule_New'
...
```

#### Additional information
These errors indicate that the generated interfaces are not linked correctly to Python3. The solution would be to adjust the cmake files that the generated python interfaces are correctly linked.

I hope I'm filing this bug in the correct repo. If not, please point me in the right direction.

Rayman, 2024-06-24:
https://github.com/ros2/rosidl_python/pull/140 changed the build to use FindPython3.cmake for building the python modules. This broke our packages that have `-Wl,--no-undefined`.

I'm not 100% sure but reading https://gitlab.kitware.com/cmake/cmake/-/issues/18100 it seems that python modules should not be linked to a python library, but instead use the symbols from the interpreter they are run from. This means the have undefined symbols by definition. 

So do we want to:
1. target_link_libraries the messages to Python3::Python which resolves the error
2. Fix all packages that have `-Wl,--no-undefined`

LarsNobleo, 2025-03-27:
I spent quite some time debugging and being completely flustered about my compilation problems before I found this issue, so this is a polite ping to start a conversation about how to solve this issue! Thanks for filing it @Rayman!

Rayman, 2025-03-27:
In my own packages I've solved this issue by putting the `-Wl,--no-undefined` after the messages have been generated. Then I can still check if my own code has any missing symbols.

LarsNobleo, 2025-03-27:
Ah sorry my last comment was not clear, that is exactly what I have done now as well. So "my issue" is solved but moving linker flags to get compiling code seems undesirable to me, so it was more to start the conversation on fundamentally fixing this, this would also ensure that no other devs encounter the same problem in the furture.